adafruit/pi_video_looper #229: 3.3v gpio trigger

lemon2007lemon, 2024-08-23:
hello all,

so complete noob here, i downloaded one of the prebuilt images and have set it up to suit my needs apart from one thing.

i can get the application to accept commands from the gpio when shorted to GND but have found that the sensors i will be using to trigger actually output 3.3v so is there a setting or could an option be implemented to trigger on pins when 3.3v is supplied?

i have messed with a few options but video_looper seems to always turn the pins high again.

again im a noob and this is my first try on the pi platform

thanks everyone

tofuSCHNITZEL, 2024-08-24:
It is no problem to change it in the python code that sets up and handles the gpio.
The gist is:
Ssh into your install, clone the repo, edit the video_lopper.py, run the install script. 
But I can see that an ini option for this would also be a good addition.

lemon2007lemon, 2024-08-24:
thank you so much for your reply

ah well that would be fantastic, as i say im completely new to this, could you please me an idea what to change in video_looper.py to achieve the needed result.

massive thanks!

tofuSCHNITZEL, 2024-08-25:
I have implemented an ini setting in the next branch
https://github.com/adafruit/pi_video_looper/tree/next

you can clone the next branch to your pi and run the install - would be great if you can test if it works and then I will release it on the main branch

lemon2007lemon, 2024-08-26:
oh wow thats great thank you.

tested and working fine, i do however have a problem with bogus triggers, tried pull up and down combinations but still having a floater issue lol

thank you so much!

tofuSCHNITZEL, 2024-08-26:
Hm what exactly are you connecting to the gpios? Do they have a common ground?

RobertJS, 2025-03-08:
Hello @tofuSCHNITZEL, your branch seems perfect to us, good job.

We use the GPIO powered with 3.3v, on pins 11 and 13 to activate video1 and video2 respectively. But, we use cables that measure several meters and connect to a relay that orchestrates 15 Raspberry (all receive the same signal). This causes us to have interferences that make the videos experience random video changes.

Is there any way to have a greater tolerance in each of the states? (3.3v / 0v) So that when we reach the ON/OFF state it maintains each state in a more stable way, more resilient to interference.

Sorry I'm inexperienced in this subject.

Greetings and thanks.

tofuSCHNITZEL, 2025-03-08:
Hi, this is not something that can be fixed in software I would say. since the voltage level that the pi accepts as "valid input" is hardware related...
maybe you need to shield the cables or add a filter capacitor:
https://raspberrypi.stackexchange.com/questions/93834/max-distance-for-gpio-cable

RobertJS, 2025-03-08:
> Hi, this is not something that can be fixed in software I would say. since the voltage level that the pi accepts as "valid input" is hardware related... maybe you need to shield the cables or add a filter capacitor: https://raspberrypi.stackexchange.com/questions/93834/max-distance-for-gpio-cable

Hello! Thank you very much for your quick response.

Could the Raspberry hardware be detecting signals just from touching the plastic of an open cable, even without touching the ends?

We are also considering modifying the internal pull-up/pull-down resistances of the Raspberry Pi 4.

Regards!

tofuSCHNITZEL, 2025-03-08:
Yes I think if the cable is not shielded anything could induce a high enough voltage to trigger the GPIO. The cable is an antenna and could pick up any kind of (radio) waves... You could try if a shielded cable would help, like cut open a STP network patch cable and connect the shield to ground. But I don't have any real world experience with long IO lines

EduardoSanchezClemente, 2025-03-08:
Hi @tofuSCHNITZEL @RobertJS,I am also suffering from the same problem and to solve it I also wanted to change the internal resistor to avoid any interference.

Although right now I am experiencing problems because when I modify the video_looper.py locally, (git add ., git commit -m....), when I restart the raspberry the code is saved with the new version but somehow the previous version of code is still running and the new changes are not applied. Additionally if you can help me, I don't know how to modify the internal resistor to make it less tolerant.

Thank you very much!!
Have a nice day!

tofuSCHNITZEL, 2025-03-09:
Git has nothing to do with the python module or code. You don't need to commit or add anything.
any changes to the packages code will not have any effect unless you reinstall the package... Or if you need to make frequent changes you could install the video looper package by adding -e to the pip install call (-e means editable so the package when run uses the "source" files in the original folder")

There is no way to modify the internal resistor of the rpi (since it's internal). You could build a separate circuit with transistors and a higher level (eg. 5v) that you then translate back down to the level of the gpios. (this will probably make everything even more complicated and error prone)