adafruit/pi_video_looper #230: Loop video until button press to move to next/prev video.

lukethemaker, 2024-09-04:
Hi.

I am making something that requires the video to be looped until you press a button to move to the next or previous video, instead of automatically moving to the next video. 

This feels like something that is possible with Omxplayer arguments, but it just feels like it would make sense to have it as a simple toggle in the INI.  It may be in there somewhere already and I'm missing it. 

First time ever using Github and I'm not sure how to mark this as feature request. I really love this project!

tofuSCHNITZEL, 2024-09-04:
I think it should be possible, but it would need some testing because the playback logic already is pretty convoluted...
Probably the best way will be, like you said, an ini option that enables "loop every file" - so all the repeat settings that you can define in the filenames are ignored.
but it will probably not be a problem hat straight forward because next/previous will not work and start the same video - which means the whole playback logic would need to be redone... So I don't think this will be possible anytime soon.

lukethemaker, 2024-09-04:
Thanks for the answer. I'll poke around a bit. For now I'll stick to opening the video files in premiere and looping them manually to make an extra long video file.

tofuSCHNITZEL, 2024-09-04:
This you can also achieve with the repeat setting in the filename https://github.com/adafruit/pi_video_looper?tab=readme-ov-file#new-in-v103 just add a very high number

closed 2025-04-01 by tofuSCHNITZEL