adafruit/pi_video_looper #232: Video not always starting and pause not always pausing

Nickman87, 2024-09-20:
- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.): https://videolooper.de/ image of raspbian on a raspberry pi zero 2.

- Python version (run `python -version` or `python3 -version`):  v3.7.3

- Error message you are receiving, including any Python exception traces:  None

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): Try to pause a video

I am using GPIO pins to trigger playing video's and I want to be able to pause them aswell so I am using the following configuration:
gpio_pin_map = "13":"video_1.mp4","11":"video_2.mp4","37":"video_3.mp4","35":"video_4.mp4","33":"video_5.mp4","31":"video_6.mp4","29":"K_SPACE"

Most of the time, my video start playing without issues, but for some reason video_5 and video_6 seem to be having issues, even though the format of the video's is all identical. I've had nothing happening, or blank screen with audio playing.
Also "pressing" pause does not seem to work every time, but I cannot seem to find a specific scenario in which this occurs.

I've looked in the runtime logs so see if there would be an issue in the GPIO not triggering, but the output shows me that the actions (play and pause) are actually being executed in the video_looper main script, so the only thing I can assume is that the communication with the omxplayer is going wrong?

What steps could I take to further debug this issue? I have ran out of options to try and get this to a more stable setup.
As when everything works, the setup is perfect, but it's not reliable right now, so I can't really put this in production.

Nickman87, 2024-09-23:
@christiansievers thank you for getting back to me so quickly.
I've further debugged with some print statement and it seems that the video not starting was mainly my fault in the changes that I made for my custom setup.

I do however have a pretty long delay on the pi zero when switching between video's, is that to be expected on this device?

christiansievers, 2024-09-24:
Hard to tell, unless you happen to have another Raspberry around. Maybe someone has more experience with this kind of setup.

closed 2024-09-28 by Nickman87