adafruit/pi_video_looper #238: Raspberry Pi A+ V1.1 for video looper?

reset131, 2025-01-26:
Is the video looper compatible with a Raspberry Pi A+ V1.1? Thank you

tofuSCHNITZEL, 2025-01-26:
yes I don't see why not.
check out the readme for the correct Raspberry Pi OS Lite image link and maybe you need to add the newer firmware files (see also readme)

closed 2025-01-26 by tofuSCHNITZEL

reset131, 2025-01-26:
thank you! So this board will be enough then?  I plan to play 1080p 25 or 50 fps.

tofuSCHNITZEL, 2025-01-26:
I honestly have no idea - never tried it - but I would think so since they have a similar GPU to 3B just a lower clock - but should be fine for 1080p30 h264.

reset131, 2025-01-26:
ok, thank you

reset131, 2025-01-26:
I will!

reset131, 2025-01-26:
btw. this is the only info I have on this particular board compatibility with the video looper, just wanted to confirm if someone did it:
"To get started you'll need a Raspberry Pi computer. Either the first generation models
(A, B, A+, B+) or the second generation Pi 2 will work great. You'll also want a USB
stick that will be used to hold movies for playback."
https://cdn-learn.adafruit.com/downloads/pdf/raspberry-pi-video-looper.pdf