adafruit/pi_video_looper #239: Option to place an Overlay for certain videos

Calvorejas, 2025-02-18:
Hi,
Still have to test it on a Pi 4, but I've been doing some research on this project and I don't know if what I want to know could be done or not, but I'll suggest it anyway, I was wondering if it would be possible to add some kind of "script" to this tool so that certain videos have an overlay on top of them. I'll elaborate my idea for a better understanding of what I'm trying to do here.

For context, I'm creating a project that aims to replicate the old TV channels from where I live, and broadcast all those channels through analog UHF signal over an antenna (All to myself, If I broadcast in a large scale, I'm sure getting into trouble haha)... I'll have a total of 9 Raspberry Pi 4 units with one 4TB drive per unit broadcasting content non stop, going all through a number of UHF modulators that broadcast each Pi 4 analog video signal to each corresponding UHF channel frequency.

I'll be using this video looper software to achive the continous flow of content I want to have with each machine. Each Pi 4 will be equal to one channel in real life.

Now that the use case is clear, lets go back to the overlay thing I was talking about... It would be nice to be able to place an overlay file, for example, a png file for the logo of the channel per video file... Why per video file and not the entire broadcast of content? Well... Because not all TV content has that logo present, for example, ads don't have the logo, or even some TV shows acquired through a channel's public archive may already have the logo on it... So it would just overlap. I don't know if what I'm suggesting is even possible... But it would be really cool to see this function being added.

Practical example:

"Antena 3 noticias 28-03-2005.mp4" (Television news) should have the file "Antena 3.png" as overlay
"Toyota Corolla 2005 ad.mp4" should not have any overlay
"The Simpsons S04E03.mkv" should have the file "Antena 3.png" as overlay

I would really love it if this could be implemented, since the alternatives would either be, not having the logo, or having to re-encode every single video I have no matter of where each file came from. And for time and archival purposes (in the case of the last one, not affecting the integrity of each file) it would be great.

I'll leave some examples attached to this "issue" (which is more like a suggestion rather than an issue).

Thanks in advance.

[image]
Broadcast without channel logo

[image]
Broadcast with channel logo

[image]
Example Overlay (resolution not precise)

tofuSCHNITZEL, 2025-03-09:
Sorry, but I don't think there is a way currently. since omxplayer takes over the frame buffer and it has no option to overlay an image.

closed 2025-03-09 by tofuSCHNITZEL

Calvorejas, 2025-03-15:
Hi @tofuSCHNITZEL 
First of all, thank you so much for reviewing my suggestion. I wasn't sure about replying here as the issue has already been changed to "closed as not planned", but in order not to create a new one just for each my inquiries while I advance with my project, I'll reply here anyways.

I have managed to rethink my strategy and, instead of using an overlay, I'm now using subtitle tracks to recreate the logo of each TV channel, and it so far seems to be working very well.

[image]
- This picture corresponds to a video test with the TV channel logo, not final, but a great approximation I did for testing.
 
 
[image]
- This picture corresponds to the very same video test, but without the TV channel logo, just for comparison. You can see here that changing the subtitle track I can hide the channel logo, this way, it is a non destructive implementation, as the original video really is untouched.
 
 
**The only problem** I'm facing right now is that, as you may vey well know, the player this project uses, omxplayer, does not support "ASS/SSA" or even "PGS" subtitles, I think it only supports SRT or the most commnon basic subtitle formats like the one I just mention... This is another limitation that this player has.

**Now my question** is, are there any plans to switch to another player like CVLC, even just normal VLC or any other player that supports this "more advance" features in the near future?

- If yes, do you have an estimation for when will this happen?
- If no, do you think it is possible for a regular user to swap omxplayer with a different player like CVLC for example without breaking the instalation too much? Can it be done? If so, how? I remember reading about this in the issues section of the project a while back, but I don't remember how the conversation ended, I remember you found issues that made the new player hard to implement, correct me if I'm wrong because I don't really know.

What I meant by breaking things, for example, excluding the obvious, like system and playback stability, I only use "advanced features" like GPIO for turning off the Raspberry Pi units I have after using them (for safety reasons), and also "sound normalization" (with a sound volume file near the rest of my video files, as the "video_looper.ini" suggests) and m3u playlists to create the structure of each TV channel's content programming (those last two features, I'm still running tests on). If those features break after the hypothetical player change, I'll have issues with the rest of the project.

In any case, don't want to make this reply any longer, thank you so much for your work and for the help.
Greetings!

tofuSCHNITZEL, 2025-03-16:
Great that it works with the subtitles (I did not know you can use images there...?)
as far as vlc is concerned - I did another test today trying to get clvc to work but its just pure horror..
just to share my frustrations:
first of it just refuses to run as root so it does not work with the whole premis of the videolooper since it needs to run as root in order to mount the usb drives. (a solution is to manipulate the root check out of the binary) - so with this hurdle done I realised that vlc takes somewhere from 1-3 seconds (to "warm up") before it displays a video... which is pretty wild in comparison to omxplayer. also looping videos is not seamless... it has quite noticable stutter in between the iterations... 

so next it looked at mpv which would work great as a replacement for omxplayer since it can loop (it stutters more than omx but less than vlc). the video almost starts instantly (not like 3 seconds of black like with vlc) - but it seems it has no support for hardware acceleration on rpi currently... which is no problem vor 1080 h264 on a pi4 but it will not play a 4k h265 smoothly. 

so if we would get mpv to handle hardware acceleartion I would say its the best replacement for omx we can get... (it seems you would need to compile a rpi optimised version of ffmpeg and then build mpv against this ffmpeg binary) but this is all to complicated for me and also not feasable to have everybody do since it takes ages to compile on an rpi... I think it would be fine to "ship" a working mpv binary with the videolooper but then - who knows if it will work on all rpis.. and it would mean a lot more testing on different rpi versions...

christiansievers, 2025-03-16:
would it help to take a look at the mp4museum project, to see how they use vlc?  https://github.com/JuliusCode/MP4MUSEUM
sorry, I’m rubbish with python.

tofuSCHNITZEL, 2025-03-16:
unfortunately it does nothing special... it suffers the same issues that VLC just has on the rpi

Calvorejas, 2025-03-18:
Sorry for the late reply... I've been researching non stop all of this but nothing comes to fruition... I am no programer at all 😅, so when it comes to code itself, I'm more lost than anything... I know more about video though...

I get your frustration @tofuSCHNITZEL It has suck that nothing so far has worked smoothly when it comes to implement VLC into the Video Looper.

In regards to the subtitles, yeah, It can be done, I ended up using a GitHub project called [IMGTOASS](https://github.com/py7hon/IMG2ASS) which takes a PNG or a graphic image file and converts it to an ASS subtitle track (which is nothing more than very small numbers, letters and symbols situated in a very specific way to try to replicate a picture)... I have to tell the program the video resolution, it's framerate and it's length. Then, as not every video player treats the subtitle tracks equally (you'll see why I say that later), I use another program called [sub2bd 0.93](https://fanrestore.com/thread-2075.html) that takes that very same ASS subtitle I previously generated and converts it to PGS (the same format used in BluRay discs) in strict mode (to force the same size and position parameters the original ASS subtitle file I previously created had) using Avisynth as it's core.

**Big problem here**. I tried playing the test videos I generated with the Windows version of MPV player and there seems to be inconsistencies with the PGS subtitles playback (and the subtitles playback in general, even SRT, which is the most basic subtitle format out there has issues)... It seems to loose sync with the track (even if there are no hicups at all in the video playback itself) and makes the subtitles disappear after some time... It sometimes is 5 minutes, but it can be more, it can be less, it deppends... If we talk about a normal subtitle track that has sporadic times to show text, with this player, the subtitles may show up, but not for the time they are intended to. ASS subtitles almost don't work with MPV, they show up, but the format is completely broken, if the ASS subtitles are set to mimic a channel logo (or any kind of picture that is just not pure text) somewhere in the screen, it ends up showing gigantic numbers and letters all over the screen because it doesn't know how to display the subtitle track propperly. This problems don't seem to be a thing with VLC.

What I'm going to say is probably absurd, if that is the case, I'm sorry, but from memory, I can remember KODI playing ASSA/PGS subtitles (and all kinds of video formats and codecs) just fine... Would not it be possible to take KODI/XBMC's video player, which I believe it's just called ["VideoPlayer"](https://kodi.wiki/view/VideoPlayer) (VideoPlayer V4L2), and implement it into the Video Looper? I also can remember it playing all kinds of videos fairly well, even 4K ones. It may be a well candidate to replace OMXplayer... That if what I'm saying makes any kind of sense (It will most likely not). KODI was using OMXplayer in the past... But switched to this one because of the changes with Raspberry Pi OS's new releases.

I don't know, I guess I'm just frustrated too... It's been months crafting this TV project myself (a lot of time and a lot of money invested in both hardware and software) and I wanted it to be as real as possible... Maybe I should just settle with burning the logos with DaVinci Resolve and call it a day. I don't know.

In any case, thank you all for your help and your work, this is the best Video Looper for Raspberry Pi I've found.