adafruit/pi_video_looper #240: Disable Sccreensaver

adtbm, 2025-02-25:
Hi, 
i am using the Videolooper and the setup was quite easy.
What i have noticed (and please exuse me if this is a simple task), the looper goes into black screen after some time. I've already looked into the .ini file, but haven't found an entry there.
It looks like, the screensaver or power save kicks in and disables the looper.
Is there an easy way to disable the Screensaver/powerSaver ?

Thank you

adtbm, 2025-02-25:
Thanks for the quick answer.
i followed the instructions in the ReadMe, i don't have a desktop at all, the looper boots and starts the videos (so i suppose it must be the Lite version).
I'll check for the Supervisor log.

adtbm, 2025-03-02:
OK, i checked again.
The Screen blacks out more or less always after running  for about 20  minutes.
I've tried in  Raspi-config to disable screen blackout, but no change.
It certainly did not crash, because once i hit ESC after the screen goes  black, i am back with normal command prompt.

tofuSCHNITZEL, 2025-03-02:
Okay weird, I have never experienced this issue before... Is there anything in the log files or in dmesg?

christiansievers, 2025-03-02:
Just a quick idea, can you try a different screen? In case it’s the screen that doesn’t play along.