adafruit/pi_video_looper #241: Freezing Image & Video Player

DerrickBarra, 2025-03-20:
**Issue**
When running overnight, on multiple occasions I've come back to see the pi_video_looper frozen.

Occurs with image_player, and omx_player.

When using image_player, I can remove the USB drive from the pi once its frozen, and the image remains on the screen.
When using omx_player, over time the gap between video loops takes longer and longer until it finally freezes after a number of loops. 

**OS & Version**
Unsure of Buster Lite version & video looper version as I started this project last year
I'll try these tests again with the latest image on videolooper.de to double check

**Configuration**
- Keyboard is disabled, as I read that can improve memory usage
- Loading media from USB in read-only mode
- OS is set to read only mode (to prevent corruption when power is cut)
- Media is set to 1024x600 resolution (matching my screen)

**Hardware**
Waveshare 7" Screen <[Link](https://a.co/d/b1LFd0u)>
Raspberry Pi Zero 2W <[Link](https://a.co/d/eF52WsQ)>
Eastbull 4gb Micro-SD Card <[Link](https://a.co/d/i396Jj1)>
Teamgroup 32GB Fat32 USB Drive <[Link](https://a.co/d/7D1UjYU)> with loaded media

**Areas To Investigate**
- Memory Leak
- Overheating

**Notes**
I haven't enabled the console_output, but I'm not sure if that would be enough to detect a memory leak or the temperature of the pi to catch an overheating issue. I wanted to report these issues in case others have any ideas while I continue to investigate.

tofuSCHNITZEL, 2025-03-20:
Sorry to hear you have issues.
I have around 30 videolooper pis running constantly for more than 3 years and also pis that are powercut daily without read only file system - never had a single issue.
I don't think that the read only helps against corruption because the problem is mostly that the power to the SD card controller is lost not that "important" files are written (and then corrupted)
so I would suggest you switch to usb copy mode to rule out a faulty usb drive (or try another drive) - and or maybe the issue is with the SD-Card of your Raspberry - maybe try a fresh/different SDcard.

cheers

DerrickBarra, 2025-03-20:
Thank you for the quick reply, I'll try ruling out hardware issues since you've been having it work consistently. Appreciate it!

christiansievers, 2025-03-20:
Also try another power supply. Some have failed on me after running for a long time - but I agree, never the SD cards. :)

DerrickBarra, 2025-03-21:
I've recreated my OS image starting from videolooper.de's latest image (v2.9), which uses video_looper v1.0.19

Changes Made From OS Image
- Disabled Startup Screens
- Enabled Splash Screen
- Enabled OS Read Only Mode (I'm loading media from USB)

I'll let the video player run overnight for a test and see if I encounter the same crashing issue. 

I'm still using the same hardware as I'm waiting on some official Sandisk USB & Micro-SD cards to arrive. But this will hopefully test if it was an issue with my video looper installation or Buster Lite Pi OS installation.